mikeryan/crackle #5: decrypting pcapfile

omarsohail, 2015-11-19:
hello mike
I'm trying to run a captured pcap file in crackle which contains all the necessary packets needed to decrypt but it still gives an error and does not detect any packets. any help would be great thanks
[image]

[image]
[image]

mikeryan, 2015-11-19:
What command line parameters did you give to ubertooth-btle to capture these packets?

omarsohail, 2015-11-19:
Ubertooth-btle -f -r /tmp/pcapfile

mikeryan, 2015-11-20:
Ah, crackle doesn't support that PCAP format (yet). You can either recapture using -c instead of -r, or wait for me to implement it. I plan to bang it out in the next few days.

omarsohail, 2015-11-20:
Ah I see, in a bit of a tight schedule so I tried using -c as 'ubertooth -f -c /tmp/pipe' or '/tmp/pipe.cap' but it doesn't seem to do anything and takes me back to the command list. thanks for being patient with me as I am still a noob.

mikeryan, 2015-11-20:
Can you paste the output of ubertooth-btle -h? Also what version of libbtbb and ubertooth are you using? What platform are you on?

omarsohail, 2015-11-20:
I'm currently using the 2015-09-R2 version on libbtbb and ubertooth one on kali linux 2.0

[image]

mikeryan, 2015-11-20:
And the output of ubertooth-btle -h?

omarsohail, 2015-11-20:
sorry wrong image
[image]

mikeryan, 2015-11-20:
It looks like libbtbb was built without PCAP support. Did you build this yourself? If so, can you apt-get install libpcap-dev and rebuild libbtbb and ubertooth (including rerunning cmake in a fresh build dir)?

omarsohail, 2015-11-20:
hey I have now updated both the firmwares and I am now getting -c but when I'm running ubertooth -f -c tmp/pipe it runs fine but no file is being saved.
[image]

mikeryan, 2015-11-20:
Does /tmp/pipe exist? Have you tried any different filenames? This is very odd behavior, I've never seen it.

omarsohail, 2015-11-20:
I managed to get it working, wireshark was being a pain in the ass and wasn't accepting pipe as a interface. -c is working fine now, Thank you very much mike!
also just wanted to ask if there are devices that are best for sniffing and decrypting?

mikeryan, 2015-11-20:
I'm a big fan of TI's BLE SensorTag. You can use BlueZ to talk to it and enable encryption, which makes it great for doing lots of trial runs.

omarsohail, 2015-11-20:
hey mike
I got crackle working and captured and entire pairing conversation between a vivosmart activity tracker and an iPhone using -c but crackle still gives me this

[image]

mikeryan, 2015-11-20:
That is a common issue: Ubertooth doesn't have a 100% packet capture rate, and it missed some of the important packets. Try re-running the pairing process several times, and you will eventually capture all of the packets. It's fairly labor intensive but it does ultimately work in the long run.

JelteF, 2015-11-27:
My change in pull request #6 makes this a bit better, since it only needs 1 confirm packet instead of two.

mikeryan, 2015-12-22:
A ton of stuff has been merged that should improve this. Give that a try, and reopen the issue if you're still having trouble.

closed 2015-12-22 by mikeryan