mikeryan/crackle #18: issue with crackle

michaelp123, 2016-03-20:
Hello,
When using crackle I am always getting the following:

No connect packet found
No pairing request found
No pairing response found
No confirm values found, at least one is needed
Not enough random values found (0, need 2)
No LL_ENC_REQ found
No LL_ENC_RSP found
Giving up due to 7 errors

I sniffed the pairing process several times with ubertooth-btle and always get this result.

What could be the issue here?
TNX.

jacky035, 2016-04-16:
I have the same error. Can help someone?

Santy99ab, 2016-05-18:
I think this is your problem: https://github.com/mikeryan/crackle/issues/5 (same as the resolved).
You must capture the packets into a .pcapng
And be sure that you have all the packets, sometimes ubertooth doesn't capture all...

thamtnguyen, 2016-06-03:
Hi guys, 
Did you solve your problem?

I got same error with crackle. Though I can see the connect request packet captured in my .pcap file, but the error is as following: 

No connect packet found
No confirm values found (0, need 2)
Not enough random values found (1, need 2)
Giving up due to 3 errors.

Any advices please?
Thanks

jsradhe, 2016-06-15:
i also have same problem.. plz try to solve this issue..

root@osboxes:~# crackle -i soundpot.pcapng -o decrypted.pcap
No connect packet found
No pairing request found
No pairing response found
Not enough confirm values found (0, need 2)
Not enough random values found (0, need 2)
No LL_ENC_REQ found
No LL_ENC_RSP found
Giving up due to 7 errors

mikeryan, 2016-06-19:
@jsradhe you need to run ```ubertooth-btle -f```, not ```-p```.

@thamtnguyen can you please attach your PCAP file?

thamtnguyen, 2016-06-19:
@mikeryan here is the PACP file, that I captured the pairing between my iPhone 6 and Apple watch.
[test_AppleWatch_2.pcap.zip](https://github.com/mikeryan/crackle/files/322553/test_AppleWatch_2.pcap.zip)

Thanks

mikeryan, 2016-06-20:
@thamtnguyen running the latest git master version of crackle against your PCAP file and got the following errors:

```
No confirm values found, at least one is needed
Not enough random values found (1, need 2)
Giving up due to 2 errors
```

This indicates that Ubertooth was unable to capture all the packets needed to crack the pairing. This unfortunately is somewhat expected behavior as Ubertooth does not have a 100% capture rate. In this scenario you must repeatedly pair and unpair, capturing each time until crackle does not give any errors such as the above.

thamtnguyen, 2016-06-20:
In fact, I tried many times capturing the pairing process by using
Ubertooth. However, I got the same errors...
I am trying the capturing with BLE Friend hardware. I will share you the
results.

Many thanks.

Best Regards

Tham Nguyen
PhD student
School of Computing and Communications
Faculty of Engineering and Information Technology
University of Technology Sydney, Australia

thamtnguyen, 2016-06-20:
In fact, I tried many times capturing the pairing process by using
Ubertooth. However, I got the same errors...
I am trying the capturing with BLE Friend hardware. I will share you the
results.

Many thanks.

12424642, 2016-08-18:
I ran ubetooth-btle -f -r  as well as -c both give me the following error

connect packet found
No pairing request found
No pairing response found
Not enough confirm values found (0, need 2)
Not enough random values found (0, need 2)
No LL_ENC_REQ found
No LL_ENC_RSP found
Giving up due to 7 errors

These are the captures;
[captures.zip](https://github.com/mikeryan/crackle/files/424835/captures.zip)

Please help @mikeryan 

Thanks in advance!

12424642, 2016-08-18:
@jsradhe Were you successful running your capture?

12424642, 2016-08-18:
I tried crackle again with no success. This was a different file that I was sure I had enough data captured in. Got the exact the same error.
[capture2.zip](https://github.com/mikeryan/crackle/files/425717/capture2.zip)

Santy99ab, 2016-09-01:
In a useful capture you will find these packets:

Connection packets:
https://imageshack.us/i/poFb1eZup

Pairing packets; Feature exchange, TK and STK generation:
https://imageshack.us/i/pmOiJgqpp

Finally, the link is encrypted with STK:
https://imageshack.us/i/pld7ehbZp

In these screenshots of wireshark  you can see all the packets: 

- connect packet found
- Pairing_Req: No pairing request found
- Pairing_Rsp: No pairing response found
- Pairing Confirm: Not enough confirm values found (0, need 2)
- Pairing Random: Not enough random values found (0, need 2)
- No LL_ENC_REQ found
- No LL_ENC_RSP found

mikeryan, 2016-11-05:
Hello all, I've just pushed a major update to the code that should help explain some of the errors you're seeing. Please pull from master and try running that against your capture files. If you're still having problems with the new code, please reopen individual tickets for your issues.

closed 2016-11-05 by mikeryan