mikeryan/crackle #19: decrypt with LTK

Santy99ab, 2016-05-18:
I'm trying to decrypt a connexion using the ltk cracked before...
But the command line show me that error:
-----------------------------------------------------------------------------------------------------------------------------
root@K4Li:~/Software/crackle-master# ./crackle -i Capturas/Prueba.pcapng -o Capturas/PruebaRecogidaDatos_dec.pcapng -l 7530d81af7aab41da8b1c2500faf6504 -v 
Violación de segmento
-----------------------------------------------------------------------------------------------------------------------------
"Violación de segmento" is like segmentation fault, and it's supposed to be caused by crackle...
¿Some help? 
Thanks.

damienkilgannon, 2016-05-24:
I am seeing the same segmentation fault .... any ideas what is happening?

Santy99ab, 2016-05-24:
In line 644 of crackle.c there are a bug:

    while ((opt = getopt(argc, argv, "i:o:svthl")) != -1) { ...

You must add ":" after l parameter "l:" (like the input "i:" or the output "o:"), basically change it into: 

    while ((opt = getopt(argc, argv, "i:o:svthl:")) != -1) {

Now the program will take the argument -l (ltk) and won't raise the segmentation fault...

This solves the argument error, but i have changed another things of the code to be able to decrtypt with ltk.

closed 2016-05-24 by Santy99ab