mikeryan/crackle #20: Sample Data

cmagorian, 2016-08-20:
What was the source of the sample data?

When dumping decrypted pcaps and the original encrypted pcaps, one sees the exact same data.

What should the result of the decrypted data look like?

Thanks!
Chris

12424642, 2016-08-20:
There isn't much data contained within the examples from what I've seen. Once you crack the 'ltk_exachange' file and compare, the only difference I've seen is that the decrypted one shows the ltk on wireshark, you should be able to see this by searching for 'encryption' in the string finder. Also the l2caps are not longer fragmented after the encryption begins. I hope this helps. Bare in mind, I'm still a bit of a noob to this so I could be wrong.

mikeryan, 2016-11-07:
@cmagorian grab the latest code from master and check out the ```tests``` subdirectory. The sample data can be found in ```tests/01_crack``` and ```tests/02_ltk_decrypt```. Each directory has a README.md that explains what's going on, and in the ```out``` subdirectory is expected output in PCAP format. You can compare those files to the input files to see the changes.

Briefly: the encrypted data will look like L2CAP fragments (as in the screenshot below). When the data is decrypted, it will look like ATT protocol PDUs and Link Layer control PDUs.

[image]

closed 2016-11-07 by mikeryan