mikeryan/crackle #21: CONNECT_REQ 0x05 and Randomized Tx Address bit: True

martijnls, 2016-08-31:
Hello, I am trying to use crackle to decrypt but have encountered a problem. When running crackle it indicates one error that no connect packet has been found, while I have found this in my capture file.

With wireshark I have found that the packet header is 0x2245 (PDU Type: CONNECT_REQ and TxAddress: True)

In de crackle.c code i have found 

//connect packet, grab those addresses!
if ((flags % 0xf) == 5)

I believe the above code should be changed as below, too test if the CONNECT_REQ (0x05) bits are set.
if ((flags & 0xf) == 5)

After changing this, the state->connect_found is marked as 1.

The crackle output is now:
TK not found, the connection is probably using OOB pairing
Sorry dood :(

Crackle does not seem to handle the "Randomized Tx Address bit: true", but is it possible to decrypt and is crackle just missing some additional handling?

martijnls, 2016-08-31:
I see this problem was

martijnls, 2016-08-31:
I see this problem has been fixed and was present in old source from

https://lacklustre.net/projects/crackle

crackle 0.1 source

martijnls, 2016-08-31:
Still getting output:

TK not found, the connection is probably using OOB pairing
Sorry dood :(

Pairing request is: AuthReqBonding -> MITM

mikeryan, 2016-11-05:
Give the latest code pushed to master a try.

mikeryan, 2017-03-13:
Closing this due to inactivity. Re-open it if you're still having this problem

closed 2017-03-13 by mikeryan