mikeryan/crackle #26: Found STK but couldn't crack

elgato212, 2017-03-06:
Hello :)

crackle found the correct TK but couldn't decrypt any packet:

```
Found 1 connection

Analyzing connection 0:
  59:1a:d3:2e:c4:42 (private) -> 63:22:fe:be:be:c6 (private)
  Found 286 encrypted packets
  Cracking with strategy 0, 20 bits of entropy

  !!!
  TK found: 112191
  !!!

  STK: 7d8df0b316f4571ba8d867de68c57a5b
  Decrypted 0 packets

Did not decrypt any packets, not writing a new PCAP
Done, processed 0 total packets, decrypted 0
Connection 0
  connect_found: 1
  preq_found: 1
  pres_found: 1
  confirm_found: 2
  random_found: 2
  enc_req_found: 1
  enc_rsp_found: 1
  pairing_public_key_found: 0
  pairing_dhkey_check_found: 0
  AA: af9aa3de
  IA: 59:1a:d3:2e:c4:42
  RA: 63:22:fe:be:be:c6
  IAt: 1
  RAt: 1
  PREQ: 03 03 10 0d 00 04 01
  PRES: 03 03 10 05 00 00 02
  MCONFIRM: 42 d9 20 e2 da 8c 06 5e 73 0f fe 3f 5c 9b 93 7f
  SCONFIRM: 37 2c 9d 3f 1b cb b3 99 2a 4b 16 b2 79 59 7d 00
  MRAND: 42 da f2 9d 95 26 b7 cc 2b 4d 85 5d 64 bf bc b0
  SRAND: 49 c5 aa 79 72 d8 d1 81 75 30 af 9d c0 c1 0e 86
  Rand: 00 00 00 00 00 00 00 00
  EDIV: 00 00
  SKDm: 4e 4e 2e ea 61 30 14 4d
  IVm:  97 a7 e7 d7
  SKDs: e8 8d 9b 4b 23 95 97 1f
  IVs:  8c 66 c3 4d
```
I dug a little bit and found that the code was entering this section in aes_ccm_ad():

```
if (memcmp(x, t, M) != 0) {
        return -1; // program goes here
    }
```

I'm sniffing with Adafruit BLE sniffer and use a custom script to convert to PPI. All packets appear correctly in Wireshark (CRC correct).

Any idea? I can send the PCAP to an email (don't want to make it public, it may contain sensitive data :).

Thanks for your help :)

mikeryan, 2017-03-06:
Please email it to mikeryan@lacklustre.net

elgato212, 2017-03-07:
Thanks for your quick answer! Email sent

jsantos98, 2017-03-12:
@elgato212 , could you please share the script to convert the files from adafruit sniffer? i'm currently facing a similar problem. 

Thanks

elgato212, 2017-03-13:
Hello,

Here is the Python script I use:

```
import sys
sys.path.append("/home/bla/tools/scapy-2.3.2/")
from scapy.all import *

pkt1 = rdpcap(sys.argv[1])
pkt2 = []
btle_header = "36750c000062090040674217807f0000".decode("hex")
ppi = PPI(dlt=147, len=24, notdecoded=btle_header)

for p in pkt1:
    btle = str(p)[0x11:]
    pkt2.append(ppi/Raw(load=btle))

wrpcap(sys.argv[2], pkt2)

```

mikeryan, 2017-03-13:
I'm still looking at this, and I don't see any reason that this should fail. I'd like to rule out your format conversion script as a possible source of error here. Do you have an Ubertooth, or can you share the name of the device so I can get one for testing?

jsantos98, 2017-03-13:
Thanks for the script!

In the meantime, I was able to make small changes in the code and added a very dirty but working change...
Basically added my own handler to support the adafruit sniffer (just to get the key, not sure if it will decrypt the packages):

`void packet_handler_especial(u_char *user, const struct pcap_pkthdr *h, const u_char *bytes) 
{
    crackle_state_t *state;
    state = (crackle_state_t *)user;

    state->btle_handler(state, h, bytes, 17, h->caplen);
}`

elgato212, 2017-03-14:
@mikeryan yes I have an Ubertooth but I can hardly sniff anything with it, while Adafruit sniffer works flawlessly. I'm testing a BLE smartwatch with iOS (https://www.withings.com/us/en/products/steel-hr)

elgato212, 2017-04-12:
Hello :)

Any news on this topic?
Thanks!

NadSo, 2017-06-30:
Hello Mark, I just sent you an Email

NadSo, 2017-07-02:
j@santos98 and @elgato212 I launched your python script and got the following errors:

kamdem@heberle-bosch:~/Desktop$ python helloworld.py
WARNING: No route found for IPv6 destination :: (no default route?)
Traceback (most recent call last):
  File "helloworld.py", line 6, in <module>
    pkt1 = rdpcap(sys.argv[1])
IndexError: list index out of range

I have a PCAP File from Bluefruit LE and want to convert it in order to be able to use crackle. Can you please help me?

Thank you in advance

mikeryan, 2020-12-06:
Closing this due to inactivity, please reopen if you're still having the same problem.

closed 2020-12-06 by mikeryan