mikeryan/crackle #27: Support for pcap files generated by Nordic sniffer

jsantos98, 2017-03-12:
It would be great to support these files.

They use dlt 157 (User 10) and payload nordic_ble

This is the device I'm using:
https://www.adafruit.com/product/2269

If necessary, I can provide some captured files.

Thanks!

mikeryan, 2017-03-13:
This has been on the roadmap for a while now. Thanks for filing the issue, I'll keep it open and close it once this feature is completed. No ETA at the moment though.

manopaul, 2017-10-12:
Mike. Any update on supporting pcapng files captured using the Nordic Sniffer. 

Getting the following error
Frames inside PCAP file not supported ! dlt_name=(null)
Frames format supported:
 [256] BLUETOOTH_LE_LL_WITH_PHDR
 [192] PPI

Thank you. 

[image]

tynbendad, 2017-10-17:
this may be the code to decode the adafruit/nordic btle packet: https://github.com/adafruit/Adafruit_BLESniffer_Python/blob/master/wireshark_dissector_source/packet-nordic_ble.c

wireshark does handle the pcap file if you change dlt=157 to use protocol=btle... but, i'm not sure what that does, compared to what this tool needs...

drneuronerd, 2018-05-10:
has there been any work done on support for Nordic Sniffer?  I see that @pccr10001 created a branch that added detection, but when I use the LTK (obtained via the process outlined [here](https://superuser.com/questions/229930/finding-bluetooth-link-key-in-windows-7-to-double-pair-a-device-on-dualboot-com) It won't decrypt the packets.  Any input would be much appreciated.

manopaul, 2018-05-10:
Unfortunately I did not get any support on this. mike must be very busy.

jgardner-r7, 2019-01-23:
Just ran into this problem using a Nordic-based sniffer...  Would be happy to share some captures if it would help.  I really want to see crackle do its thing with this stuff I'm looking at :unlock:

mikeryan, 2019-01-23:
If you can attach some sample captures to this thread that would be perfect.

jadware, 2019-03-04:
here's a sample file that shows a similar error: http://s000.tinyupload.com/index.php?file_id=16588735506536596039

ikarus23, 2019-08-15:
This is most likely a duplicate of https://github.com/mikeryan/crackle/issues/35 since the Adafruit BLE sniffer "Bluefruit LE Sniffer" uses the NRF sniffer firmware. I've created the pull request https://github.com/mikeryan/crackle/pull/44 which gives makes crackle support PCAP files generated by the NRF sniffer. I've tried it on a BLE keyboard and was able to recover the LTK. I've not tested if decryption works.

ikarus23, 2019-08-15:
Ok, maybe this is off topic, but when I sniff the pairing procedure of an BLE keyboard with the Nordic sniffer, I get "Encrypted packet decrypted incorrectly (bad MIC)" for each packet after pairing (except for Empty PDUs) using wireshark. According to Nordic, the sniffer should decrypt the packets. Also, when I save the dump, crackle says that there are no packets to decrypt in this dump.

ikarus23, 2019-08-20:
Some news: I missed something in the output. Crackle does tell me that there are encrypted packets and it does tell me the PassKey. But it does not show me the LTK and it says "Decrypted 0 packets".

Ubertooth:
```
Analyzing connection 19:
  cc:3d:82:36:cd:3c (public) -> e4:0c:4f:54:55:98 (random)
  Found 48 encrypted packets
  Cracking with strategy 0, 20 bits of entropy

  !!!
  TK found: 969341
  !!!

  Decrypted 48 packets
  LTK found: 7a7bb213d072e7b488f25d9f8b4f78da
```

Nordic Sniffer (Adafruit board):
```
Analyzing connection 1:
  fb:87:02:ec:bd:66 (random) -> e4:0c:4f:7b:55:98 (random)
  Found 284 encrypted packets
  Cracking with strategy 0, 20 bits of entropy

  !!!
  TK found: 576239
  !!!

  Decrypted 0 packets
```

I guess there is something wrong the the Nordic sniffer, the way it saves the data to the pcap or the way the packets are handled by crackle. As I said, I had no time so far to look into the packet handler. I've just copied it from @pccr10001.

closed 2020-12-06 by mikeryan