mikeryan/crackle #29: Issues in finding pairing

sgript, 2017-06-26:
Hi there,

So I've been trying to use crackle to find and decrypt a pairing between my Pebble smartwatch and iPhone. Here's the pcap attached:

[Uploading capture.pcap.zip…]()

I used Ubertooth's command to try catch a pairing: `ubertooth-btle -f -c capture.pcap`

However when running `crackle -i capture.pcap -o foo.pcap`, I get this issue:
```
No connect packet found
No pairing request found
No pairing response found
Not enough confirm values found (0, need 2)
Not enough random values found (0, need 2)
No LL_ENC_REQ found
No LL_ENC_RSP found
Giving up due to 7 errors
```

If anyone could share some insight for this as I'm still beginning, would be very helpful.

titanick, 2017-08-03:
your command is right ,but you shuld know that ubertooth default channel is 37 ，but ble default channel is 37 38 39 ,so your packet is not right .

Kay887, 2017-08-09:
Kay887 commented Aug 9, 2017
Hi Mike,

I have the same issue. I am able to see the LL_ENC_REQ and the pairing messages in Wiresharl but crackle gives me the error that there are no requests found. So do I keep changing the advertising channel to a 37,38 or a 39 till I get the right packet

Kay887, 2017-08-09:
I am sorry. I am relatively new to this. Would really appreciate some help.

Kay887, 2017-08-10:
Hi Mike,

I have the same issue. I am able to see the LL_ENC_REQ and the pairing messages in Wiresharl but crackle gives me the error that there are no requests found. So do I keep changing the advertising channel to a 37,38 or a 39 till I get the right packet? I am sorry. I am relatively new to this. Would really appreciate some help.

mikeryan, 2017-08-10:
@sgript your PCAP contains no pairing packets. Refer to the FAQ entry: https://github.com/mikeryan/crackle/blob/master/FAQ.md#crackle-is-complaining-about-missing-packets-why-cant-i-crack

@Kay887 can you attach a PCAP?

Kay887, 2017-08-10:
@mikeryan Ive attached six different PCAPs recorded at different channels through the command: ubertooth-btle -f -A 37 -c test1.pcap and for all channels. 
[Files.zip](https://github.com/mikeryan/crackle/files/1214874/Files.zip)

mikeryan, 2017-08-10:
Using crackle master I am able to crack several of the key exchanges in your files. The following files work:

```37/test12.pcap```
```38/test1.pcap```

Make sure you're using the master branch from GitHub

Kay887, 2017-08-11:
Thanks for the reply. When you say master branch of GitHub, could you refer me to the page. I am kinda newbie to this. Would really appreciate some guidance. 

Thanks for the help again

Kay887, 2017-08-11:
Thanks for the reply. So I should download the version on the master branch of GitHub?. Cool. Thanks. Will give it a try and let you know.

mikeryan, 2017-08-11:
```
git clone https://github.com/mikeryan/crackle.git
cd crackle
make
./crackle -i <your_file.pcap>
```

Kay887, 2017-08-11:
Thanks for the quick reply. Will give it a try!!!! :)

Kay887, 2017-08-11:
Thank you so so very much for the help. It worked!!!!! 

Appreciate it!!!!

Kay887, 2017-08-11:
Just wanted to ask you one question:
While I was decrypting the packets, I found different LTKs for different communication.  I wanted to know does the LTK keep changing for every communication? If I unpair the device and pair them again, would there be a different LTK generated for that specific communication or will it be the same for different communication?

Sorry for silly question

mikeryan, 2017-08-11:
Every time you pair, a new LTK is established. After you've done that, you can use encrypted communication with a key based on that LTK without pairing again. If you do pair again, both devices get new LTKs.

Kay887, 2017-08-18:
@mikeryan 

I watched one of your videos I guess it was Shmoocon where in you have talked about injection using ubertooth to faux slave mode. I tried doing that, the tx led glows very lightly but when I perform hcitool lescan, for some reason the fake mac address does not reflect in the list. It does show the BLE device I am using but not the mac address created using ubertooth. 

Another thing I wanted to ask you is when you talked about packet injection, is it possible to inject BLE packets into an ongoing communication between 2 BLE devices?

Thanks

mikeryan, 2017-08-18:
The faux slave code is very crappy and I'm not surprised it doesn't work well. It doesn't do anything useful even when it does work.

We do not have any code for usefully injecting packets into BLE connections, or generating any true Bluetooth traffic to speak of.

Kay887, 2017-08-20:
Thanks for the reply. Just one last thing, what is the ubertooth-tx used for?

NadSo, 2017-08-26:
Hello Mike I sent you alot of Email but you didn't reply.
I have my Android App talking to my BLE (BTLC1000) I am able to light the LED on and off via BLE. Now I want to fuzz my BLE Device. You said in one of your presentation that it is possible to do that with PyBT if I got it right. But the thing is there is not documentation about PyBT. Can you please help me or give me some tipps?
Thank you

closed 2018-03-18 by sgript