mikeryan/crackle #31: Encrypted packets found but not decrypted

rajibeee, 2018-01-12:
It is not showing any errors but not decrypting the connections as well. What's wrong ?

```
crackle -i captured.pcapng -o decrypted-dlt_PPI_without-LTK.pcap
Found 1 connection

Analyzing connection 0:
  52:59:91:36:91:31 (random) -> f0:c7:7f:fc:da:8e (public)
  Found 12 encrypted packets
  Unable to crack due to the following error:
    Missing both Mrand and Srand

Did not decrypt any packets, not writing a new PCAP
Done, processed 0 total packets, decrypted 0
```

Then tried with LTK

```
rajib@rajib-lab-desk:~/bluetooth$ crackle -i captured.pcapng -o decrypted-dlt_PPI-LTK.pcap -l e77655c8523f02fc93c05e2fa279f1b8
Found 1 connection

Analyzing connection 0:
  52:59:91:36:91:31 (random) -> f0:c7:7f:fc:da:8e (public)
  Found 12 encrypted packets
  Decrypted 0 packets

Did not decrypt any packets, not writing a new PCAP
Done, processed 0 total packets, decrypted 0
```

mikeryan, 2018-01-12:
For the first case, you're missing the required packets to crack the
encryption key:

https://github.com/mikeryan/crackle/blob/master/FAQ.md#crackle-is-complaining-about-missing-packets-why-cant-i-crack

In the second case, it appears that the LTK is not correct. Where did
you get that LTK?

mikeryan, 2018-01-12:
Can you share a screenshot of the log file that shows it?

rajibeee, 2018-01-12:
[image]

mikeryan, 2018-01-12:
Thanks, one last thing: can you also share the .pcap file that you're
using with crackle?

rajibeee, 2018-01-12:
remove the .zip at the end of the file name

[captured-using-c-Multiple-LL_ENC_REQ.pcap.zip](https://github.com/mikeryan/crackle/files/1627436/captured-using-c-Multiple-LL_ENC_REQ.pcap.zip)
[captured-using-r.pcapng.zip](https://github.com/mikeryan/crackle/files/1627437/captured-using-r.pcapng.zip)

"captured using" is used in the name to identify which ubertooth command I used to capture.
I used this command when using `-c`
`ubertooth-btle -f -tF0:C7:7F:FC:DA:8E -c captured-using-c.pcap`

with the _captured-using-c_ file I am getting the following error

```
crackle -i captured-using-c.pcap -l e77655c8523f02fc93c05e2fa279f1b8 -o decrypted-from-c.pcap
Warning: found multiple LL_ENC_REQ, only using latest one
Found 1 connection

Analyzing connection 0:
  54:0f:3b:31:0d:28 (random) -> 47:e8:cd:9b:bd:d8 (random)
  Found 0 encrypted packets
  Decrypted 0 packets

Did not decrypt any packets, not writing a new PCAP
Done, processed 0 total packets, decrypted 0
```

Danyc0, 2020-03-13:
Hi @rajibeee, I know it's been a while but I think I've solved your issue, the LTK you provide is in the incorrect byte order for Crackle. Crackle takes it in Big-Endian, whist some sniffers will output the LTK in Little-Endian.

See my successful run with your input file:
[image]

And the output file (just remove the .zip from the end):
[decrypted-from-c.pcap.zip](https://github.com/mikeryan/crackle/files/4329498/decrypted-from-c.pcap.zip)

closed 2020-12-06 by mikeryan

Mahmoud-Rezk, 2021-11-04:
@rajibeee can you told me the way which you got the LTK key number with cause I have the same problem

rajibeee, 2021-11-05:
> @rajibeee can you told me the way which you got the LTK key number with cause I have the same problem

Used wireshark.

You have to use the LTK in the opposite order that you get from Wireshark to properly decrypt it.

Mahmoud-Rezk, 2021-11-07:
@rajibeee  
Nut how with Wireshark the LTK is not shared on the air to get it from Wireshark.

rajibeee, 2021-11-08:
> @rajibeee but how with Wireshark the LTK is not shared on the air to get it from Wireshark.

I do not know about the latest BLE version. As far as I remember I got this on 4.1 or 4.2. It was almost 4 years ago.