mikeryan/crackle #32: Problems with decrypting packets from BT Smart keyboard

D93, 2018-01-12:
Hi! Unfortunately I'm getting trouble with decoding packets from my BTLE Smart keyboard by Microsoft. I'm sniffing with an Ubertooth One (Firmware version: 2017-03-R2 (API:1.02), ubertooth 2017-03-R2).

My approach: 

> Pairing the keyboard with a laptop, writing anything (=smashing one key several times and hoping to find it anywhere in the packets later), unpairing it again, rinse repeat.

I aim to show that it is possible to get some interesting information (text/passwords/...) by sniffing BTLE keyboard packets. 
Assuming that the packets send by a keyboard are encrypted, I try to use crackle. Unlickily I do not understand what I am doing wrong.

```
root@kali:~# crackle -i microsoft_btle_kb_multiple_pairings.pcapng -o crackle_output.pcap
Found 2 connections

Analyzing connection 0:
  WINDOWS_LAPTOP_MAC (public) -> KEYBOARD_MAC (random)
  Found 0 encrypted packets
  Unable to crack due to the following errors:
    Missing both Mrand and Srand
    Missing LL_ENC_REQ
    Missing LL_ENC_RSP

Analyzing connection 1:
  WINDOWS_LAPTOP_MAC (public) -> KEYBOARD_MAC (random)
  Found 0 encrypted packets
  Unable to crack due to the following errors:
    Missing both Mrand and Srand
    Missing LL_ENC_REQ
    Missing LL_ENC_RSP

Did not decrypt any packets, not writing a new PCAP
Done, processed 0 total packets, decrypted 0

```

I'd appreciate any help.

mikeryan, 2018-01-12:
This is covered in the FAQ:

https://github.com/mikeryan/crackle/blob/master/FAQ.md#crackle-is-complaining-about-missing-packets-why-cant-i-crack

The short answer is that you'll need to keep trying. You might have
slightly more luck with some experimental Ubertooth firmware available
in the le_phy tree on GitHub.

https://github.com/greatscottgadgets/ubertooth/tree/le_phy

D93, 2018-01-13:
> The short answer is that you'll need to keep trying.
I hate to say it, but until now I coulnd't achieve any results with lots of trying. I'll continue to try though, even if I'm not optimistic about this anymore.

> You might have slightly more luck with some experimental Ubertooth firmware available in the le_phy tree on GitHub.
I might try it as my last resort before giving up on this one.
So I cloned this branch and ran _make_ in the "firmware" directory. 
But how do I flash it on a Ubertooth? Normally one can find a "ubertooth-firmware-bin" directory with a flashing script in it, but not in this case.

mikeryan, 2018-01-13:
Go into firmware/bluetooth_rxtx, run make in there, and then flash using
ubertooth-dfu -r -d bluetooth_rxtx.dfu

D93, 2018-01-23:
Still getting 
```
Missing both Mrand and Srand
    Missing LL_ENC_REQ
    Missing LL_ENC_RSP
```
but I don't think it's an issue of _crackle_ - my packet dumps just don't contain all of the needed packets.

closed 2018-01-23 by D93