mikeryan/crackle #35: cracking BLE packets captured through Adafruit Bluefruit LE Sniffer

rickypan108, 2018-03-28:
hello,

I captured BLE packets (including pairing information) by using Adafruit Bluefruit LE Sniffer which uses the Nordic's NRF sniffer software. I can see all the pairing related information in wireshark but when i run the packet capture through crackle , crackle is unable to read/interpret the packets. Everything works fine when i use Ubertooth to capture the BLE packets that include paring information and i can crack the link layer encyrption using crackle in that case. I noticed that unlike ubertooth , in case of Adafruit/nordic packet capture  analysis on wireshark , i need to add DLT_User (157) as "nordic_ble"  ( it is DLT_User (147) > btle in case of ubertooth).

Can you advise if there is a simple way so that i could get crackle to read packets captured through Adafruit Bluefruit LE Sniffer?

Thanks
Rick

ikarus23, 2019-08-12:
https://github.com/mikeryan/crackle/pull/36

ikarus23, 2019-08-12:
I've updated the pull request to be compatible with the latest version of crackle and the Nordic BLE sniffer. Check out https://github.com/mikeryan/crackle/pull/44 or use [my fork](https://github.com/ikarus23/crackle) in the meantime.

ikarus23, 2019-08-15:
This is most likely a duplicate of https://github.com/mikeryan/crackle/issues/27 since the Adafruit BLE sniffer "Bluefruit LE Sniffer" uses the NRF sniffer firmware.

llcoder00, 2020-08-19:
pccr10001 didn't work for me using Nordic nRF52840-DK

closed 2020-12-06 by mikeryan