mikeryan/crackle #38: Unable to decrypt

Shashank6669, 2018-05-25:
Hi Mike,
I'm unable to decrypt my bluetooth captures. I end up getting the following error even though the LL_ENC_REQ, LL_ENC_RSP, and LL_START_ENC_REQ have been captured.

Error: 
Unable to crack due to the following error:
    Missing both Mrand and Srand

 I really appreciate your work and your help would be invaluable to me. I've attached my pcap files.
[pcap.zip](https://github.com/mikeryan/crackle/files/2037828/pcap.zip)

Thanks!

mikeryan, 2018-05-25:
Refer to the last FAQ entry: https://github.com/mikeryan/crackle/blob/master/FAQ.md#crackle-is-complaining-about-missing-packets-why-cant-i-crack

The Mrand and Srand packets are required to calculate the keys used during pairing. You'll have to retry your capture repeatedly until you capture those required packets.

Shashank6669, 2018-05-25:
Thanks a lot @mikeryan ! I got it decrypted after several attempts. Will be able to decrypt future exchanges now that I can gain access to the LTK? Also do you know where the LTK would be in the decrypted pcap file? I just started learning about Bluetooth protocol and I apologize if my question is too silly!
Thanks!
[decrypted.zip](https://github.com/mikeryan/crackle/files/2037995/decrypted.zip)

mikeryan, 2018-05-25:
Crackle will output the LTK to the terminal. You can use it to decrypt future connections as long as you don't re-pair.

Shashank6669, 2018-05-25:
@mikeryan , I did not get the LTK output in the terminal. I just got the TK as 0.  I've attached the screenshot of my output.

[image]
Thanks!

mikeryan, 2018-05-25:
Looking at the decrypted output with the Wireshark filter "btsmp", it appears you didn't capture the packet in which the LTK was exchanged. I'm afraid you'll need to keep making attempts to recover that value.

It's worth confirming that you actually need to sniff and crack to achieve your goal. What are your devices and what are you ultimately trying to achieve?

Shashank6669, 2018-05-25:
The output I shared was obtained from cracking this specific pcap file:
[pairing6GOLD.zip](https://github.com/mikeryan/crackle/files/2040486/pairing6GOLD.zip)

I used the "btsmp" filter as you suggested and have I think the LTK was exchanged.
[image]

My devices are a samsung network AP and a Linux machine/Android device. A Bluetooth connection is established to share network parameters. I'm trying to sniff the packets and extract the data.

Thanks!

mikeryan, 2018-05-25:
Refer to the first and second FAQ entries: https://github.com/mikeryan/crackle/blob/master/FAQ.md#can-i-log-data-without-having-to-sniff

You can log data in Linux before it is encrypted using btmon. You can do the same in Android using HCI Packet Logging. You will not need Ubertooth or crackle in these cases.

Danyc0, 2020-03-13:
Just to answer one of your other questions @Shashank6669 (once you get a complete and decrypted sniff) the LTK(Long Term Key) will be in an SMP Encryption Information packet, see my sniff below:
[image]

closed 2020-12-06 by mikeryan

itewqq, 2021-07-26:
> Just to answer one of your other questions @Shashank6669 (once you get a complete and decrypted sniff) the LTK(Long Term Key) will be in an SMP Encryption Information packet, see my sniff below:
> [image]

@Danyc0 Hi there, I encountered a similar problem with @Shashank6669 when using crackle, and I also found LTK in your picture. But I used the LTK here to try to decrypt the two devices that have been paired afterwards but failed, with the "Did not decrypt any packets" message.

In addition, I also used another LTK similar to package 5549 in your picture, but the decryption still failed. 

I also tried the reverse order of bytes, but it still failed.

I am very confused about what is going on, and would appreciate it if you can provide some explanation, thank you!

/cc @mikeryan @Shashank6669