mikeryan/crackle #41: Crackle can't find TK

FrancescoTaurino, 2018-12-14:
Hello,

I am using crackle for recovering the TK. I am using a pcap file generated by Btlejack with the command
`btlejack -c any -x ll_phdr -o btlejack_cap.pcap` and it contains all the necessary packet (CONNECT_REQ, Pairing Request and Response, Pairing Confirms and Pairing Randoms). 

Then with crackle, I run the command `crackle -i btlejack_cap.pcap` and the output is ( I remove th device addresses):
```
Warning: No output file specified. Decrypted packets will be lost to the ether.
Found 1 connection

Analyzing connection 0:
  xx:xx:xx:xx:xx:xx (public) -> yy:yy:yy:yy:yy:yy (public)
  Found 98 encrypted packets
  Cracking with strategy 0, 20 bits of entropy
    TK is not found. The connection could be using OOB pairing or something
    else fishy is going on. File a bug with more info about the devices.
    Sorry d00d :(

Done, processed 0 total packets, decrypted 0
```

The pairing was done via Passkey Entry (not JustWorks or OOB) and BLE v4.1.

Comments and suggestions are welcome!  Thank you.

mikeryan, 2018-12-14:
Can you include the PCAP and the actual TK value if you know it?

FrancescoTaurino, 2018-12-18:
The problem was in Btlejack which sets in Wireshark the address type of the Initiator and the Advertiser to public when they actually are random.

closed 2020-12-06 by mikeryan