mikeryan/crackle #43: Crackle: Invalid interface capture length

Op4ik, 2019-04-29:
Hi.
Output:
When I use pcapng:
_crackle: invalid interface capture length 4294967295, bigger than maximum of 262144_
If I use pcap, then crackle can't find anything.
Output:
_Warning: unknown packet type encountered, skipping
Warning: unknown packet type encountered, skipping
No connect packet found
No pairing request found
No pairing response found
Not enough confirm values found (0, need 2)
Not enough random values found (0, need 2)
No LL_ENC_REQ found
No LL_ENC_RSP found
Giving up due to 7 errors_

Sniffer: 
CC2642R1 with default SmartRF software

Op4ik, 2019-04-29:
[Pcaps.zip](https://github.com/mikeryan/crackle/files/3127895/Pcaps.zip)

Danyc0, 2020-03-13:
If you look in the PCAPs you uploaded, they're UDP data, not Bluetooth data, are you sure you're not getting your files mixed up? Where did you get your PCAPs from?

mikeryan, 2020-12-06:
This does indeed look like a case of @Op4ik supplying a non-Bluetooth file to crackle. Please reopen if you're still having trouble.

closed 2020-12-06 by mikeryan