mikeryan/crackle #52: Found TK, STK but 0 packets decrypted

chungchungdev, 2023-05-06:
I came across a problem exactly like this [closed issue](https://github.com/mikeryan/crackle/issues/26).

I am using the [nRF sniffer](https://www.nordicsemi.com/Products/Development-tools/nrf-sniffer-for-bluetooth-le)(3rd-party nRF52840 based hardware) with wireshark to get the packets.

```
$ crackle -v -i mydevice_pairing2.pcapng -o decryted.pcapng
PCAP contains [NORDIC_BLE] frames
Warning: packet is too short to be encrypted (1), skipping
Warning: packet is too short to be encrypted (1), skipping
Found 1 connection

Analyzing connection 0:
  mac1 (random) -> mac2 (random)
  Found 2179 encrypted packets
  Cracking with strategy 0, 20 bits of entropy

  !!!
  TK found: 6digits
  !!!

  STK: 5e0be96c4f80c856e9aba58917126545
  Decrypted 0 packets

Did not decrypt any packets, not writing a new PCAP
Done, processed 0 total packets, decrypted 0
Connection 0
  connect_found: 1
  preq_found: 1
  pres_found: 1
  confirm_found: 2
  random_found: 2
  enc_req_found: 1
  enc_rsp_found: 1
  pairing_public_key_found: 0
  pairing_dhkey_check_found: 0
  AA: af9a852b
  IA: mac1
  RA: mac2
  IAt: 1
  RAt: 1
  PREQ: 0b 0b 10 2d 00 04 01
  PRES: 03 02 10 05 00 00 02
  MCONFIRM: 6c a3 32 c4 7f ae c3 b8 77 ac 86 37 f0 d3 48 b4
  SCONFIRM: c8 72 56 6c 4f 26 d1 01 7d 58 b5 9d 4b 6c 0b 13
  MRAND: 39 3c d8 24 06 ce 15 2a 7c 28 60 10 fe 28 dc 6c
  SRAND: 14 6f 12 c2 ad b8 c0 9b 24 20 b4 cf f3 41 96 21
  Rand: 00 00 00 00 00 00 00 00
  EDIV: 00 00
  SKDm: 0e 60 16 0f 5a ae 4a 99
  IVm:  31 57 96 8d
  SKDs: c0 81 bf 1c 50 4c a6 e6
  IVs:  18 7c 29 54
```

I can send some files through email if you are interested.

chungchungdev, 2025-03-03:
Thanks for the reply. Back then I was intercepting the data between my BLE device and smartphone. It turned out that I can use the Bluetooth HCI log from smartphone directly to read the data. But it is necessary to do the pairing every time to get the decrypted data.

closed 2025-03-03 by chungchungdev

libcrack, 2025-03-04:
Are you saying that you had to do the pairing every time with your Android device to be able to get the decrypted data?

May I ask if did you capture the (1)pairing process + (2)key exchange + (3)encrypted connection between your Android phone and the BLE device using the nRF52840 dongle + nRF Sniffer? or did you just used the Snoop HCI Log from your Android phone?

The interesting thing about the nRF52840 dongle + nRF Sniffer is to actually sniff third-party BLE connections, but I'm having the very same problems that you suffered: TK found, STK found but 0 decrypted packets (therefore no LTK).

Thanks for the update! :-)

chungchungdev, 2025-03-04:
I just use the HCI log. The HCI log is good enough for me so I didn't dive into the sniffing approach.

libcrack, 2025-03-04:
Ahhhh fair enough. Thanks for the info mate! :-)