mikeryan/crackle #53: Decrypt LE Secure Connection with known LTK does not work

jkrauss1987, 2023-11-22:
Hi, I have a pcap file from sniffing an LE Secure owner pairing. I have got the correct LTK on another way. 
The output of the tool reports a successful decription and a new output.pcap file is generated. But when i load it into Wireshark the payload of the decypted packets is much less than in the original files messages and Wireshark says "bad MIC" just like in the original file. 

Original File: 
[image]

Decrypted File:
[image]

So my question is: Is the "decrypt with LTK" part of crackle able to decript LE Secure Connection when the LTK is known?   

Many Thanks!

jkrauss1987, 2023-11-22:
I just saw that the decryption seems to work well but the problem is rather that the BT LE Link Layer and the BTL2CAP Layer are not recognized/parsed by Wireshark. I think that might be because the BTLE.CRC value is not recalculated but instead set to 0x 00 00 00. Is there a way to say Wireshark to ignore that? 
Regards

trentp-igor, 2024-01-05:
Decrypting LE Secure Connection with LTK doesn't seem to work at all for me.

Without supplying LTK, it fails as expected:
```
 Analyzing connection 0:
  44:01:bb:a0:d6:17 (public) -> 48:27:e2:2d:9a:66 (public)
  Found 23 encrypted packets
  Unable to crack due to the following error:
    LE Secure Connections
```

With LTK it just doesn't decrypt the packets for some reason:
```
[crackle]$ ./crackle -i ../pairing.pcapng -l AD1FE6CCEE0E20E701108DE0141BE9ED -o test.pcap
…
Analyzing connection 0:
  44:01:bb:a0:d6:17 (public) -> 48:27:e2:2d:9a:66 (public)
  Found 23 encrypted packets
  Decrypted 0 packets
```

XenoKovah, 2025-02-04:
@trentp-igor FWIW it worked for me on a Just-works Secure Connection when I put the endianness correct. I.e. currently Wireshark displays stuff little-endian, so if I just copy the value from Wireshark it doesn't work. But when I reversed the order of bytes to the big-endian the tool expects, it worked. Given that you're posting in caps and Wireshark doesn't use caps I'm guessing you didn't get the LTK from Wireshark, but still worth giving it a try to reverse the order.

XenoKovah, 2025-02-04:
@jkrauss1987 given that you confirmed decryption works w/ LTK shouldn't you close this as actually working and just open a ticket for Wireshark people to ask how to ignore invalid CRC? (Which it sounds like was a problem from the source capture, not CrackLE. Perhaps use Sniffle instead, since it calculates CRCs correctly?)

closed 2025-02-05 by jkrauss1987